PXL-Embedded-AI/repo #25: bestellingslijst doorsturen:

Villadelfia, 2020-03-11:
- link
 - productcode
 - prijs
 - aantal
 - bij 3 leveranciers

Doorsturen ten laatste 12 maart avond.

i5cP7i, 2020-03-11:
De (low-cost) diffraction grating cards zijn te verkrijgen bij Amazon via Rainbow symphony. De "Vincent leermiddelen Scientific" [leermiddelen.be](http://www.leermiddelen.be/nl/dias-en-filters) biedt een hogere kwaliteit aan voor een hogere prijs, en meerdere relevante artikelen (zoals diafragma's, filters, halogeenlampen, etc.). 

Bijgevoegd de (voorlopige) bestellingslijst:

[onderdelen-EmbeddedAI-v1_00.pdf](https://github.com/PXL-Embedded-AI/repo/files/4321415/onderdelen-EmbeddedAI-v1_00.pdf)
[onderdelen-EmbeddedAI-v1_00.docx](https://github.com/PXL-Embedded-AI/repo/files/4321417/onderdelen-EmbeddedAI-v1_00.docx)

i5cP7i, 2020-03-11:
Update:
[onderdelen-EmbeddedAI-v1_01.pdf](https://github.com/PXL-Embedded-AI/repo/files/4321548/onderdelen-EmbeddedAI-v1_01.pdf)
[onderdelen-EmbeddedAI-v1_01.docx](https://github.com/PXL-Embedded-AI/repo/files/4321549/onderdelen-EmbeddedAI-v1_01.docx)

Villadelfia, 2020-03-18:
Doorsturen naar Koen. Bart & Vincent in CC.

Villadelfia, 2020-03-18:
Daarna closen en in done zetten.

i5cP7i, 2020-03-18:
Bestellingslijst is verzonden naar Koen Gilissen:

[image]

closed 2020-03-18 by i5cP7i