PXL-Embedded-AI/repo #26: Flowchart data transfer

kiks3107, 2020-03-25:
Maak een flowchart van hoe de data overgebracht word tussen de teams

kiks3107, 2020-03-29:
https://www.lucidchart.com/invitations/accept/0c2d74eb-7f20-437c-937b-e89df13227c4
link naar de flowchart die ik heb gemaakt

Villadelfia, 2020-03-29:
Screenshot maken en _in de repository uploaden._ Ik ga geen account aanmaken op een random website, en de docenten gaan dat ook niet willen.

pxltech, 2020-03-29:
Klopt.

Mvg
Vincent

[http://www.pxl.be/Assets/afbeeldingen_algemeen/MailFootersPXL/Footer%20infodag%2025_04_2020_210.jpg]

kiks3107, 2020-03-29:
sorry ik dacht dat met die link iedereen er aan kon zonder een account te moeten maken

i5cP7i, 2020-05-05:
Updated flowchart (sinds 28 april). Zie Flowchart/flowchart-embeddedAI.png

[Link naar flowchart.](https://github.com/PXL-Embedded-AI/repo/blob/master/Flowchart/flowchart-embeddedAI.png)